enjoy-digital/litex #2305: How to replace PLIC to APLIC?

looongbin, 2025-08-09:
Hello,
    I finished APLIC which interface is similar to the PLIC in Vexiiriscv. Now I want to replace PLIC in Litex with APLIC to boot with Vexiiriscv. But I couldn't find a universal PLIC call logic, can give me some suggestion about that?

Dolu1990, 2025-08-09:
Hi,
Currently the VexiiRiscv SoC infrastructure is in SpinalHDL (PLIC / CLINT), so you would need to swap it there : 
https://github.com/SpinalHDL/VexiiRiscv/blob/dev/src/main/scala/vexiiriscv/soc/litex/Soc.scala#L233

Dolu1990, 2025-08-09:
Ahhh so for the software ?

=> you would need to modify https://github.com/enjoy-digital/litex/blob/master/litex/soc/cores/cpu/vexiiriscv/irq.h

inochisa, 2025-08-09:
> Ahhh so for the software ?
> 
> => you would need to modify https://github.com/enjoy-digital/litex/blob/master/litex/soc/cores/cpu/vexiiriscv/irq.h

Good, It seems like we can add a new `__riscv_aplic__` for it.

closed 2025-08-13 by looongbin