enjoy-digital/litex #2321: neorv32 fails to build (multiple assignments for "addr_mask_napot" , "addr_mask" )

Ruinland, 2025-09-01:
Hi Florent,

I just bumped into this error while trying to build neorv32 for orange crab.

It gives out following error : 
```
/home/ruinland/machdyne_lakritz/lol_root/litex/litex/soc/cores/cpu/neorv32/neorv32_cpu_pmp.vhd:90:10:error: multiple assignments for "addr_mask_napot" offsets 0:239
  signal addr_mask_napot, addr_mask : addr_mask_t;
         ^
/home/ruinland/machdyne_lakritz/lol_root/litex/litex/soc/cores/cpu/neorv32/neorv32_cpu_pmp.vhd:90:27:error: multiple assignments for "addr_mask" offsets 0:239
  signal addr_mask_napot, addr_mask : addr_mask_t;

...

Traceback (most recent call last):
  File "/home/ruinland/machdyne_lakritz/lol_root/neorv32_test/../litex-boards/litex_boards/targets/gsd_orangecrab.py", line 231, in <module>
    main()
    ~~~~^^
  File "/home/ruinland/machdyne_lakritz/lol_root/neorv32_test/../litex-boards/litex_boards/targets/gsd_orangecrab.py", line 224, in main
    builder.build(**parser.toolchain_argdict)
    ~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/ruinland/machdyne_lakritz/lol_root/litex/litex/soc/integration/builder.py", line 373, in build
    self.soc.finalize()
    ~~~~~~~~~~~~~~~~~^^
  File "/home/ruinland/machdyne_lakritz/lol_root/litex/litex/soc/integration/soc.py", line 1505, in finalize
    Module.finalize(self)
    ~~~~~~~~~~~~~~~^^^^^^
  File "/home/ruinland/machdyne_lakritz/lol_root/lol_venv/lib/python3.13/site-packages/migen/fhdl/module.py", line 156, in finalize
    subfragments = self._collect_submodules()
  File "/home/ruinland/machdyne_lakritz/lol_root/lol_venv/lib/python3.13/site-packages/migen/fhdl/module.py", line 149, in _collect_submodules
    r.append((name, submodule.get_fragment()))
                    ~~~~~~~~~~~~~~~~~~~~~~^^
  File "/home/ruinland/machdyne_lakritz/lol_root/lol_venv/lib/python3.13/site-packages/migen/fhdl/module.py", line 102, in get_fragment
    self.finalize()
    ~~~~~~~~~~~~~^^
  File "/home/ruinland/machdyne_lakritz/lol_root/lol_venv/lib/python3.13/site-packages/migen/fhdl/module.py", line 156, in finalize
    subfragments = self._collect_submodules()
  File "/home/ruinland/machdyne_lakritz/lol_root/lol_venv/lib/python3.13/site-packages/migen/fhdl/module.py", line 149, in _collect_submodules
    r.append((name, submodule.get_fragment()))
                    ~~~~~~~~~~~~~~~~~~~~~~^^
  File "/home/ruinland/machdyne_lakritz/lol_root/lol_venv/lib/python3.13/site-packages/migen/fhdl/module.py", line 102, in get_fragment
    self.finalize()
    ~~~~~~~~~~~~~^^
  File "/home/ruinland/machdyne_lakritz/lol_root/lol_venv/lib/python3.13/site-packages/migen/fhdl/module.py", line 157, in finalize
    self.do_finalize(*args, **kwargs)
    ~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^
  File "/home/ruinland/machdyne_lakritz/lol_root/litex/litex/build/vhd2v_converter.py", line 215, in do_finalize

```

The command I am using is 
`../litex-boards/litex_boards/targets/gsd_orangecrab.py --cpu-type=neorv32 --cpu-variant=full+debug --cpu-cfu=true --build`

LiteX components version : `2025.4`

ghdl version : 
```
GHDL 6.0.0-dev (5.1.1.r39.g68548a981) [Dunoon edition]
 Compiled with GNAT Version: 15.2.1 20250813
 static elaboration, mcode JIT code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2025 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

yosys : 
```
 /----------------------------------------------------------------------------\
 |  yosys -- Yosys Open SYnthesis Suite                                       |
 |  Copyright (C) 2012 - 2025  Claire Xenia Wolf <claire@yosyshq.com>         |
 |  Distributed under an ISC-like license, type "license" to see terms        |
 \----------------------------------------------------------------------------/
 Yosys 0.55+109 (git sha1 2223d7848, g++ 15.1.1 -march=x86-64 -mtune=generic -O2 -fstack-protector-strong -fno-plt -fPIC -O3)
```

stnolting, 2025-09-06:
Hey @Ruinland.

Which version of the core are you using?

> `/home/ruinland/machdyne_lakritz/lol_root/litex/litex/soc/cores/cpu/neorv32/neorv32_cpu_pmp.vhd:90:10:error: multiple assignments for "addr_mask_napot" offsets 0:239`

This looks familiar... I think we fixed this bug (or at least a similar one) quite some time ago: https://github.com/stnolting/neorv32/pull/1236

stnolting, 2025-09-09:
> However, the latest one seems still triggering the same problem.

Hm, that's strange... Could you test your setup with the modified PMP file (https://github.com/stnolting/neorv32/blob/dev090925/rtl/core/neorv32_cpu_pmp.vhd) from https://github.com/stnolting/neorv32/tree/dev090925?